dsa110/dsa110-issues #6: dsa110master failed to boot after power cycle.

rh-codebase, 2020-02-22:
The power in Bldg12 was turned off for A/C installation. After power was restored dsa110master failed to boot while trying to mount external USB disk. The system doesn't like the entry in /etc/fstab. 

Workaround:
* comment out /etc/fstab entry for USB disk
* boot system
* manually mount USB disk

TODO:
* Fix /etc/fstab entry.

@danasimard Do you know anything about the fstab entry?

danasimard, 2020-02-24:
@rh-codebase I added that manually.  Do you have an idea of what is wrong with it?  I hadn't tested rebooting lxd100master but I believe it worked with `mount --all`

danasimard, 2020-03-12:
I edited the fstab entry and tested further with mount -a

closed 2020-03-12 by danasimard