dsa110/dsa110-issues #9: etcd watch callback failure

caseyjlaw, 2020-03-31:
Repo: dsa110-pyutils
Branch: cjl/dev

I am using the python etcd3 client to watch a key for changes. I add the callback function (just prints the event dictionary) like this:
```
de.add_watch('/mon/ant/{0}'.format(antnum), my_cb)
```

When the callback function is called, I get this error:
```
Watch callback failed
Traceback (most recent call last):
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/etcd3/watch.py", line 223, in _safe_callback
    callback(response_or_err)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/dsa110_pyutils-v2.0.1_rc1_1_g6250ff8_dirty-py3.6.egg/dsautils/dsa_store.py", line 167, in a
    key = event.key.decode('utf-8')
AttributeError: 'WatchResponse' object has no attribute 'key'
```

Am I missing something about how to do a callback?

caseyjlaw, 2020-04-01:
My latest commit now pushed (https://github.com/dsa110/dsa110-pyutils/commit/e256cbabec194bff750f45f434ade0ee50c7b7df). Fixed a silly bug, but that then revealed this other issue cited here.

rh-codebase, 2020-04-01:
What are you using for antnum? I'll add some protection around this for starters.
What version of dsa_store are you running against? It should be v2.0.1-rc1.

caseyjlaw, 2020-04-01:
I tried antnums of 8, 26.
I am using that version.

rh-codebase, 2020-04-02:
No you're not. Thank goodness python prints this out in the trace(I didn't think it did). Look at the stack trace. You're calling against: dsa110_pyutils-v2.0.1_rc1_1_g6250ff8_dirty-py3.6.egg.

rh-codebase, 2020-04-02:
Please describe a method to reproduce. I tried the following:

caseyjlaw, 2020-04-02:
I log in to dsamaster, activate my anaconda environment (python3.6), install the latest from my branch (`git checkout cjl/dev`, `python setup.py install`), then:

```
(py36) claw@dsamaster:~$ dsamon watchant 26
Traceback (most recent call last):
  File "/home/claw/anaconda3/envs/py36/bin/dsamon", line 11, in <module>
    load_entry_point('dsa110-pyutils===v2.0.1-rc1-1-g6250ff8-dirty', 'console_scripts', 'dsamon')()
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/dsa110_pyutils-v2.0.1_rc1_1_g6250ff8_dirty-py3.6.egg/dsautils/cli.py", line 47, in watchant
    vv = de.get_dict('/mon/ant/{0}'.format(antnum))
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/dsa110_pyutils-v2.0.1_rc1_1_g6250ff8_dirty-py3.6.egg/dsautils/dsa_store.py", line 127, in get_dict
    return json.loads(data.decode("utf-8"))
AttributeError: 'NoneType' object has no attribute 'decode'
```

rh-codebase, 2020-04-02:
It's unclear when this changed, but the event passed into the callback is a WatchResponse object which contains a list of events in an attribute named 'events'. For watching a specific key, this list is 1 element so the proper syntax for extracting the key,value in a some_callback(event) function is event.events[0].key.decode('utf-8') and event.events[0].value.decode('utf-8').  Fixing in next release.

rh-codebase, 2020-04-02:
Fixed at v2.0.2-rc1 (ef49fa1)

closed 2020-04-02 by rh-codebase