dsa110/dsa110-issues #10: dsa110-pyutils etcd query should catch non-existent antenna values

caseyjlaw, 2020-04-15:
An issue for recent-ish versions of pyutils and looking at antenna data with dsa_store. In this case, I'm looking with the python CLI, but I think this is an issue upstream of the CLI.

```
> dsamon ant 111
Traceback (most recent call last):
  File "/home/claw/anaconda3/envs/py36/bin/dsamon", line 11, in <module>
    load_entry_point('dsa110-pyutils===v2.0.2-rc1-6-g1ddc046-dirty', 'console_scripts', 'dsamon')()
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/dsa110_pyutils-v2.0.2_rc1_6_g1ddc046_dirty-py3.6.egg/dsautils/cli.py", line 24, in ant
    vv = de.get_dict('/mon/ant/{0}'.format(antnum))
  File "/home/claw/anaconda3/envs/py36/lib/python3.6/site-packages/dsa110_pyutils-v2.0.2_rc1_6_g1ddc046_dirty-py3.6.egg/dsautils/dsa_store.py", line 127, in get_dict
    return json.loads(data.decode("utf-8"))
AttributeError: 'NoneType' object has no attribute 'decode'
```

rh-codebase, 2020-04-20:
If the class can detect non-existent keys, then it it could throw a KeyDoesNotExistException. Implementation could also provide an API to return an array of keys below a given parent key. @jwl-cit was looking for something like this. Propose: subkeys(parent_key: String) -> Array. An empty array if no subkeys are found. KeyDoesNotExistException if parent_key doesn't exist.

caseyjlaw, 2020-04-20:
That sounds like a nice feature.

rh-codebase, 2020-04-20:
Etcd key API moved to #13. This issue remains to implement KeyDoesNotExistException.

caseyjlaw, 2020-04-22:
I have this implemented in the cjl/dev. Shall I do a PR to merge it (and other small change) into development?

rh-codebase, 2020-04-23:
I don't think there's a need to do a pull request. Did you pull in the latest on development into your cjl/dev with this latest feature and test? If so, you can push your branch up and I'm happy to take a look. If you've pulled in the latest development, then updating development to your feature will just be a fast-forward.  When I updated master and development, I thought I left a dangling tag which can be cleaned up by a fetch --prune.  My view of the tree looks clean(meaning I see only 2 lines of dev)

caseyjlaw, 2020-04-23:
I've merged development into cjl/dev and pushed that to github. The merge had a small conflict on .gitignore, but that was simple to fix.
The cli.py module now has a way to try/except when getting antenna info with `dsamon ant <antnum>`. Once the key check feature is implemented, I can test better.