dsa110/dsa110-issues #12: Interactive array monitoring map

rh-codebase, 2020-04-20:
Morgan brought up an enhancement for an interactive array monitoring map where once could click on an antenna to view monitor points. This display would work well with the alert feature to show which antennas had monitor points outside their range. The linux nagios package has a very similar interface for displaying computer systems and switches which also includes tree-like  inheritance. Nagios uses the  snmp protocol I believe.

mcatha, 2020-04-20:
I wonder if this could be somehow integrated with Google Earth or Maps.  I think you would want it to be zoomable, because at the scale where you can see the whole of OVRO, it would be difficult to differentiate the Tee antennas.

rh-codebase, 2020-04-20:
Nagios is a symbolic tree presentation so it solves the problem of scale. Spatial information can be handy though especially if some transformer fails and the display shows a group of antennas being offline.