dsa110/dsa110-issues #16: array and deployment notes

caseyjlaw, 2020-07-13:
The components of the array and their deployment are sometimes managed in general "notes". These should be free form text that can be entered by anyone, but associated with a time and author. They should be visible to all users to assist the planning of observations and interpretation of array performance.

Sources of notes:
1. webUI field
2. command-line tool

Display of notes:
1. webUI
2. command-line tool

Rick can do webUI integration and Casey can do the command-line tool.

Qs:
* Is this all stored in etcd? If so, define format in mp spreadsheet.

rh-codebase, 2020-07-13:
My understanding was this was an interface into syslog  not etcd. If integrating these notes only with application logging, then syslog is sufficient but if needing to compare against monitor data then etcd -> influxdb is likely desired as well. If interfaced into influxdb, then we should probably set a limit on message size and whether it can contain UTF-8 characters or just plain ASCII.

caseyjlaw, 2020-07-13:
Oh, duh. Sorry, I was multitasking and forgot the design.
This is definitely a project built around syslog.

closed 2022-03-03 by caseyjlaw