dsa110/dsa110-issues #20: too many repos in dsa110-shell

VR-DSA, 2020-09-03:
there are too many repos in dsa110-shell! I suggest removing the following:
 - dsa110-alignment
 - dsa110-issues
 - DSA2000-Whitepaper-Simulation
 - dsa110-caltest?

caseyjlaw, 2020-09-11:
What happens when dsa110-shell has lots of repos? Too slow? Too much disk space?

rh-codebase, 2020-09-11:
speed and space will always be a function of repo number and repo size but if you want the entire project you have no choice but to download it all. It's pretty difficult to have a text only repo so large that it takes a long time to download but there are a few here that have data(i.e. dsa110-alignment). @VR-DSA do you really mean this shell or the .mrconfig used by the containers to provision? That's a separate file and IMO could be trimmed. As for the .mrconfig in this shell, I'd prefer to have it contain all repos since I use mr quite heavily. IIRC git attic is a framework for storing data type files in git in an attempt to get around sluggishness but I believe the first clone still takes a big hit. Another alternative is to create a .mrconfig_production type file containing only those repos for running the system. Then one could use:  'mr -c .mrconfig_production checkout' to get the result you are after. The default(without -c) would still grab all repos.