dsa110/dsa110-issues #21: pyuvdata baseline/uvw coordinates

danasimard, 2020-09-22:
For dsa110-meridian-fs

Currently meridian-fringestopping using only uvw coordinates.  pyuvdata should be able to phase properly itself if proper antenna coordinates and baseline names are included.

caseyjlaw, 2020-10-20:
Following up on this, since LWA folks are interested in pyuvdata....
Can you elaborate on the bug and whether you have an open issue with pyuvdata folks?

danasimard, 2020-10-21:
This isn't a bug in pyuvdata! I just wanted to ensure that the metadata written by the fringestopping code was consistent with the sign conventions in pyuvdata.

I've confirmed this now - the DSA correlator is doing A*B, which is consistent with the definition in pyuvdata.  The baselines are B-A, and when pyuvdata writes to a measurement set or uvfits file, it conjugates the visibilities instead changing the metadata to be consistent with the AIPS/CASA definition.

pyuvdata uses astropy to convert to ICRS frame, where it calculates uvw coordinates.  The baseline length during these coordinate conversions is only being preserved at a level of about 0.6 cm in 100 m (fractional error 5e-5, while CASA preserves the baseline length to a level of 4e-16).

closed 2020-10-21 by danasimard