dsa110/dsa110-issues #22: Build beamformed data visualization tool

caseyjlaw, 2020-09-24:
There are two libraries for reading/writing beamformed data.
-  https://github.com/UCBerkeleySETI/blimpy
- https://github.com/thepetabyteproject/your

Pick one that works for our needs and write a data quality visualization tool.

caseyjlaw, 2020-09-28:
@greghell What do you like to use for filterbank visualization?

greghell, 2020-09-28:
@caseyjlaw I simply use sigpyproc.Readers.FilReader

caseyjlaw, 2020-09-28:
Could you share an example here? Or a link to github, if available. (I was planning on writing something new, but we have a tight deadline and should reuse code where we can.)

greghell, 2020-09-28:
import sigpyproc
from sigpyproc.Readers import FilReader
import matplotlib.pyplot as plt
import numpy as np

tmp = FilReader("/home/user/vikram/scratch/fil_B0531+21_1.fil").readBlock(0,1024);
plt.figure()
plt.subplot(121);
plt.imshow(tmp,aspect='auto');plt.xlabel('frequency channel');plt.ylabel('time index');plt.colorbar();
plt.subplot(122);
plt.plot(np.mean(tmp,1));plt.grid();plt.xlabel('frequency channel');plt.ylabel('power spectral density [arb. unit]');
plt.show()

caseyjlaw, 2020-10-22:
Hi @greghell, I was reviewing our project plans and saw this issue.
Do you think the visualization of filterbank/beamformed data is done? Or do you think it would be useful to build something more general than the plotting you described above?