dsa110/dsa110-issues #31: bokehservice container setup

caseyjlaw, 2020-10-24:
After the recent restart of containers, I see bokehservice needs some more work to run bokeh antmon tool.

1. The setup for bokehservice needs to run
`conda install bokeh`

2. The new syslog version (James's new syntax) isn't tagged right. It is in development only. Is the solution to tag and merge it to main?

caseyjlaw, 2020-11-11:
@rh-codebase The bokehservice conatainer is broken again (after restart?). The annotated tagging of dsa110-pyutils is a big issue.

rh-codebase, 2020-11-11:
are referring to the repo in  bokehservice or github? In github I still don't see a version tag on your merge between origin/developmment and origin/cjl/dev. If you see our version tag locally, try pushing with --tags option.

caseyjlaw, 2020-11-11:
Ok, I added tag v2.7.0 to master, which is the same as development now. Take a look and let me know what you think.

rh-codebase, 2020-11-11:
Yikes. somewhere along the development path, the tests broke. README.md states to:

rh-codebase, 2020-11-11:
Yep. my environment. I'll update the README to state to be sure to pip install . at TOT before running test. All tests pass. at v2.7.0 for me. Doesn't look like the Documentation has been updated(running sphinx) and commit.

rh-codebase, 2020-11-12:
bumped version: v2.7.1 includes updated Docs and test results.

rh-codebase, 2020-11-12:
Got sidetracked with pyutils. The original issue was the need to run:
conda install bokeh

which still needs to be added to the bokeh container profile.

caseyjlaw, 2020-11-12:
I actually added a reply to another ticket about that, I think. But yeah, we can close two of we can get a chance to define special classes of container.

rh-codebase, 2020-11-12:
Ah. Nice. We can close this then with a ref.to that issue. There must be a way to add a linked ref to another issue.

caseyjlaw, 2020-11-13:
Special containers will be dealt with in issue #32

caseyjlaw, 2021-01-04:
Sorry, but I need to reopen this issue. I found that bokeh wasn't running in the container after a recent restart.
The bokehservice container needs one more thing to restart automatically. Currently, I need to do this:
`nohup bokeh serve antmon.py &`

But could that be done automatically? Or would you prefer to use serviced?