dsa110/dsa110-issues #32: corr21 container setup

caseyjlaw, 2020-10-26:
The corr21 container needs the following run by default.
- `pip install hdbscan seaborn`

rh-codebase, 2020-10-26:
All corrXX containers are identical. That is, they all use the same profile. If the other containers have this package installed, then there was a provisioning error. If this is a new package not captured in the current provisioning scripts, I can add it but it will apply to all corrXX containers. If there's a need to have distinct corrXX profiles, I can create an issue for that feature.

caseyjlaw, 2020-10-26:
Maybe the latter is best. We know that there are two kinds of corr containers. One for correlator and another for search code. Maybe we could make two flavors that are similar, but allow the search-like ones to have hdbscan and seaborn (and others that we'll think of later).

rh-codebase, 2020-10-27:
How many 'search' containers are needed for the production system?

caseyjlaw, 2020-10-27:
IIRC, corr21 and maybe one or two others numbered similarly will be dedicated to search. Each may have one container of this flavor.

rh-codebase, 2020-10-27:
Do you mean hosts as in lxd110h21 and possibly 2 others will be dedicated to search? The hXX hosts run the containers. Now it's possible to have containers run containers but I don't see any real benefit to this paradigm as it will likely induce more confusion and there's no performance increase. How about we reserve 15min this Wed. to discuss the architecture options.The 'search' sounds like a nice use case to explore the possibilities. Some topics to discuss(or fill me in on)
- Should 'search' live in separate containers or be processes running on existing containers?
- What communication requirements does search need?
- Which networks will handle this communication?

caseyjlaw, 2020-10-27:
Sure, we can chat more. Maybe one flavor for all the "corr"s is fine, but containers are easy to build, right?
I may have confused things by using the wrong name when saying "corr21". I suppose that is the container, not the server. I agree there is no need for containers in containers.

caseyjlaw, 2020-11-06:
We also need corr21 to the latest matplotlib (>3.0). I just ran `conda upgrade --all` and got all updated anaconda packages. 
That got what I needed, but I'm not sure if that's ok for others on corr21. If so, then it would be good to have the container deploy with this anaconda command, too.

caseyjlaw, 2020-11-11:
I'll move a comment I made from another ticket here...
* The bokehservice container will not work after reprovisioning because it needs an extra `conda install bokeh`.
* This ticket mentions another issue which is that corr21 needs an extra `pip install hdbscan seaborn` and `conda upgrade --all`.

Do we have a way of making those containers do that by default (during provisioning)?