dsa110/dsa110-issues #34: systemtd unable to restart antenna monitor control process

rh-codebase, 2020-11-02:
systemd will attempt to restart a failed process a fixed number of tries. Here is what it captured when trying to restart the antenna monitor and control process:

rh-codebase, 2020-11-02:
On closer look. systemd is reporting that the process is still active and running, so the exception it reports is not stopping the process.

 **antmc.service - DSA110 Antenna MC Service
   Loaded: loaded (/etc/systemd/system/antmc.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-10-30 21:38:08 UTC; 3 days ago**

jwl-cit, 2020-11-02:
I believe that the communications with the LabJacks was interrupted somehow and the LabJack drivers (ljm) were not able to reconnect. The main process keeps running but the individual LabJack threads crash. The main process does not keep track of the threads, but this could be implemented. This will be on the backburner for a while because of other priorities, though.

rh-codebase, 2020-11-02:
Might be a network issue. Do you have a time when the Labjacks lost connectivity? If not, I should be able to back it out from the monitor data. Vikram reported losing connection with corr00 for about a minute a few days ago. That's on a different switch so might be pointing to the host(lxd110maas) or the container(dsa110maas) for the MaaS software. I'll see what I can dig up on those. Can you tell if connectivity was lost to all jabjacks, or just one(a few)?

jwl-cit, 2020-11-02:
All LabJacks (Ant and BEB)
(casa) ubuntu@antservice:~$ grep "Exception in thread" /var/log/syslog
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb31:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb51:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb91:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb61:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb1:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb111:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb101:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb81:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb11:
Nov  2 11:50:38 antservice start_hwmc.py[18933]: Exception in thread beb21:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread beb71:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread beb41:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant18:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant27:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant26:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant19:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant30:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant14:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant33:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant34:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant29:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant116:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant28:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant20:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant17:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant15:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant31:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant16:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant101:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant13:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant35:
Nov  2 11:50:39 antservice start_hwmc.py[18933]: Exception in thread ant100:

jwl-cit, 2020-11-02:
I don't know what time zone the time stamps are written in.

rh-codebase, 2020-11-02:
UTC.  Thanks for those times. On the host machine(lxd110maas), running the antservice container, the logs show(there's a hostname problem with this host as it keeps resetting its name to dsa110maas. It really is lxd110maas!!):

**dsa@dsa110maas:~$ grep "Nov  2 11:50" /var/log/syslog
Nov  2 11:50:32 dsa110maas systemd-networkd[2004]: enp5s0: Lost carrier
Nov  2 11:50:32 dsa110maas kernel: [3502944.208056] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov  2 11:50:32 dsa110maas kernel: [3502944.208317] br1: port 1(enp5s0) entered disabled state**

In fact, the link started to misbehave quite badly over ~ 1.5 min period:

dsa@dsa110maas:~$ grep enp5s0 /var/log/syslog
Nov  2 11:50:32 dsa110maas systemd-networkd[2004]: enp5s0: Lost carrier
Nov  2 11:50:32 dsa110maas kernel: [3502944.208056] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov  2 11:50:32 dsa110maas kernel: [3502944.208317] br1: port 1(enp5s0) entered disabled state
Nov  2 11:51:11 dsa110maas kernel: [3502983.280145] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Nov  2 11:51:12 dsa110maas systemd-networkd[2004]: enp5s0: Gained carrier
Nov  2 11:51:12 dsa110maas systemd-networkd[2004]: enp5s0: Configured
Nov  2 11:51:12 dsa110maas kernel: [3502983.387853] br1: port 1(enp5s0) entered blocking state
Nov  2 11:51:12 dsa110maas kernel: [3502983.387858] br1: port 1(enp5s0) entered forwarding state
Nov  2 11:51:13 dsa110maas systemd-networkd[2004]: enp5s0: Lost carrier
Nov  2 11:51:13 dsa110maas kernel: [3502984.700040] br1: port 1(enp5s0) entered disabled state
Nov  2 11:51:16 dsa110maas systemd-networkd[2004]: enp5s0: Gained carrier
Nov  2 11:51:16 dsa110maas systemd-networkd[2004]: enp5s0: Configured
Nov  2 11:51:19 dsa110maas systemd-networkd[2004]: enp5s0: Lost carrier
Nov  2 11:51:19 dsa110maas kernel: [3502990.435723] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov  2 11:51:49 dsa110maas systemd-networkd[2004]: enp5s0: Gained carrier
Nov  2 11:51:49 dsa110maas systemd-networkd[2004]: enp5s0: Configured
Nov  2 11:51:49 dsa110maas kernel: [3503021.248171] br1: port 1(enp5s0) entered blocking state
Nov  2 11:51:49 dsa110maas kernel: [3503021.248175] br1: port 1(enp5s0) entered forwarding state
Nov  2 11:51:53 dsa110maas systemd-networkd[2004]: enp5s0: Lost carrier
Nov  2 11:51:53 dsa110maas kernel: [3503024.871478] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov  2 11:51:53 dsa110maas kernel: [3503024.871601] br1: port 1(enp5s0) entered disabled state
Nov  2 11:51:56 dsa110maas kernel: [3503028.107829] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Nov  2 11:51:56 dsa110maas systemd-networkd[2004]: enp5s0: Gained carrier
Nov  2 11:51:56 dsa110maas systemd-networkd[2004]: enp5s0: Configured

So, indeed the interface(for the ant.pvt network) went down. This could be on the switch side. I'll have to see if I can determine which side went down. The ant.pvt switch plugged into this interface is one brought down from bldg 7, a linksys LGS552). It's an old switch, far older than the server. This switch is then trunked to the Planet dual color fiber switch aggregating the antennas.

rh-codebase, 2020-11-02:
do the antenna labjacks have logs which would indicate whether their interface went down?

jwl-cit, 2020-11-02:
No, there’s no logging internally.

rh-codebase, 2020-11-11:
what's up with 11:5x?  Another connection drop to antennas and beb's. This time at 11:51:18
**(casa) ubuntu@antservice:~$ grep hwmc /var/log/syslog | grep ant | grep Except
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant15:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb21:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant28:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant17:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant18:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant24:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb61:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb51:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant25:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb1:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant20:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant30:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant14:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb91:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant102:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant35:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb81:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb41:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant26:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant32:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb11:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant31:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant116:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant19:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb111:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb31:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant33:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant29:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant27:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant100:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread ant101:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb71:
Nov 10 11:51:18 antservice start_hwmc.py[7380]: Exception in thread beb101:**

and dsa110maas indicates the inteface is down as well although a bit later:

ubuntu@dsa110maas:~$ grep enp5s0 /var/log/syslog
Nov 10 11:51:52 dsa110maas kernel: [4194220.480304] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov 10 11:51:52 dsa110maas kernel: [4194220.480608] br1: port 1(enp5s0) entered disabled state
Nov 10 11:51:55 dsa110maas kernel: [4194223.496586] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
Nov 10 11:51:58 dsa110maas kernel: [4194226.216470] br1: port 1(enp5s0) entered disabled state
Nov 10 11:52:26 dsa110maas kernel: [4194254.064436] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Nov 10 11:52:26 dsa110maas kernel: [4194254.064752] br1: port 1(enp5s0) entered blocking state
Nov 10 11:52:26 dsa110maas kernel: [4194254.064755] br1: port 1(enp5s0) entered forwarding state

Both dsa110maas and antservice run on the same physical host so time is identical for them(they share the system clock which is tied to NTP). The time different in reporting though my be due to whenever the kernel gets around to checking the interface. We can test this.

dsa-storage was added to this switch around 1500 PST today so if the entire switch goes down we should see  its interface dropping thus pointing to the switch.

rh-codebase, 2020-11-18:
Caught another connection loss to the labjacks. Since the labjacks don't keep local logs, I moved dsa-storage onto the ant.pvt network a few weeks back(++). Now this event was recorded on 2 separate machines seeming to indicate a switch issues. I doubt that cron is responsible since dsa-storage and lxd110maas system clocks are in 2 different time-zones(PST and UTC respectively) and CRON is time-zone agnostic. Following are logs from dsa-storage(time in PST) showing the ant.pvt interface dropping out:

**Nov 18 03:51:43 dsa-storage kernel: [8003565.605745] ixgbe 0000:05:00.1 enp5s0f1: NIC Link is Down
Nov 18 03:52:23 dsa-storage kernel: [8003605.609583] ixgbe 0000:05:00.1 enp5s0f1: NIC Link is Up 1 Gbps, Flow Control: None
Nov 18 03:52:29 dsa-storage kernel: [8003612.065855] ixgbe 0000:05:00.1 enp5s0f1: NIC Link is Down
Nov 18 03:53:00 dsa-storage kernel: [8003642.413950] ixgbe 0000:05:00.1 enp5s0f1: NIC Link is Up 1 Gbps, Flow Control: None**

On lxd110maas(the host running antservice, time in UTC), the logs show:

**Nov 18 11:52:24 lxd110maas kernel: [4885448.766801] br1: port 1(enp5s0) entered disabled state
Nov 18 11:52:29 lxd110maas kernel: [4885454.498526] igb 0000:05:00.0 enp5s0: igb: enp5s0 NIC Link is Down
Nov 18 11:52:29 lxd110maas kernel: [4885454.498752] br1: port 1(enp5s0) entered disabled state**

And the time the antenna mc code logged(time in UTC) the reconnect failure:

**Nov 18 11:51:49 antservice start_hwmc.py[20798]: labjack.ljm.ljm.LJMError: Address -1, LJM library error code 1239 LJME_RECONNECT_FAILED
Nov 18 11:51:49 antservice start_hwmc.py[20798]: Exception in thread ant100:**

(++) dsa-storage was also moved to allow the calibration software to have access to etcd to push cal results.