dsa110/dsa110-issues #89: Some voltage triggers not written to disk

danasimard, 2021-08-30:
Some voltage triggers are getting json header files but not being recorded properly in the dumps.dat file (in ~/data on the CORR nodes) and not written to disk.  Here is an example:
* Between start and stop of system there was one T2 voltage trigger, called 210830aaed
* That voltage trigger is not recorded in dumps.dat, but 210830aaed.json is written
* When stop.py is run, dumps.dat is updated (see below) but the trigger name is garbled.  No 210830aaed.dat is written.
* It seems part of the real trigger name was appended to the flush trigger.

Content of dumps.dat:
starting...
written trigger from specnum 100620388 TRIGNUM1 DUMPNUM0 0-flush-
0830aaed-

Grep of the logs for 210830aaed:
Aug 30 17:35:48 s_src@corr03 dsaX_trigger[38023]: control_thread: received buffer string 158148-210830aaed-
Aug 30 19:25:38 s_src@corr03 dsaX_trigger[38023]: control_thread: received buffer string 0-flush- 0830aaed-
Aug 30 19:25:47 s_src@corr03 dsaX_trigger[38023]: written trigger from specnum 100620388 TRIGNUM1 DUMPNUM0 0-flush- 0830aaed-

rh-codebase, 2021-08-30:
garbled logs could be a threading issue somewhere in the chain. If a single thread is mangling the logs, then i'd look to memory corruption in the app.

danasimard, 2021-08-30:
Ah, that's good to know.  The info saved in dumps.dat might be parsed from the logs, explaining the garbling in both placed.

I'm also unsure if the specnum here (158148) is too early for a voltage dump to be written. This may be part of the problem.  I'll keep monitoring this to see if it's a recurring issue.

rh-codebase, 2021-08-30:
It's worth checking that's for sure. It's not clear how much of this is coming from Python or C++.