dsa110/dsa110-issues #112: webserveruiservice static plot directory changed

caseyjlaw, 2022-01-13:
The webserverservice was used to share static plot (e.g., calibration png files). That was exposed as webserveruiservice:9090, which still is working after the recent reprovisioning/rebooting. However, now I can see one directory lower than before, so some files are exposed. That should be set to show only `webPLOTS`.

rh-codebase, 2022-01-13:
changed mount point on lxd110h00 to:
mount 10.40.0.208:/mnt/data/dsa110 /mnt/data

caseyjlaw, 2022-03-17:
I think this mount point is still incorrect. When loading port 9090 and the directory "/plots", then I see a bunch of other directories, including "webPLOTS". We'd like the mount point to be in that one.

rh-codebase, 2022-03-17:
h00 was accidentally powered off last Wed. when adding new disks. I think Vikram hand mounts this dir as I don't see any fstab entries and h00 is not using autofs. I'll take  a look but I'm sure this is due to the reboot.