dsa110/dsa110-issues #113: bokehservice not running bokeh server

caseyjlaw, 2022-01-13:
After the recent reprovisioning/reboot, the bokeh web server is not working. Looking on bokehservice, I see the repo is available on the server in the standard location, but the `bokeh` package is not installed. I suspect that the provisioning isn't set correctly for this container.

rh-codebase, 2022-01-24:
bokeh pkg installed: 0.13.0 but still erroring out with:

`  File "/home/ubuntu/anaconda3/envs/casa/lib/python3.6/site-packages/bokeh/protocol/receiver.py", line 8, in <module>
    from tornado.concurrent import return_future
ImportError: cannot import name 'return_future'
`

According to 
https://discourse.bokeh.org/t/importerror-cannot-import-name-return-future/3410

a newer version of bokeh is needed or a downgrade of tornado.  pip doesn't seem to know of a more recent version of bokeh. looking into it.

rh-codebase, 2022-01-24:
removed existing package with pip uninstall bokeh, then reinstall with pip install bokeh. This brought in version 2.3.3. sudo systemctl start bokeh started the service. Closing this issue and opening a new one to update provisioning.

closed 2022-01-24 by rh-codebase