dsa110/dsa110-issues #114: influx is missing t1mon and t2mon

caseyjlaw, 2022-01-13:
After reprovisioning, the t1mon and t2mon influx "measurements" are no longer available. Those were defined recently (December site visit), so they may not have been capture in the provisioning process.
We need the etcd keys under /mon/T1 and /mon/T2 ingested to influx for monitoring purposes.

rh-codebase, 2022-01-13:
which if any monitor points need to be tags?

caseyjlaw, 2022-01-13:
@liamconnor may be able to help with that.
We had configured it to create a `t1mon` and `t2mon` to be used for system health purposes. I don't recall which were tags.

caseyjlaw, 2022-01-13:
Actually, I can answer this based on an old slack converasation. I believe `t1_num` and `t2_num` are the tags.

rh-codebase, 2022-01-13:
I'm not seeing any monitor data under /mon/t1 or /mon/t2 or /mon/T1 or /mon/T2. Is this expected right now?

rh-codebase, 2022-01-14:
Following config added to etcd2dbConf.yml
`  /mon/T1:
    tablename: "t1mon"
    tagnames:
      - "t1_num"
  /mon/T2:
    tablename: "t2mon"
    tagnames:
      - "t2_num"
`
config file also updated for Provisioning.

closed 2022-01-14 by rh-codebase