dsa110/dsa110-issues #122: need to add "statusmon": a new influx measurement

caseyjlaw, 2022-01-29:
The etcd key `/mon/status/1` is set up to be a new influx measurement `statusmon`. This will help track system health.
The key payload has a field `time` in mjd and `status_num`, which is the tagged field (this should be identical structure to `t1mon` and `t2mon`, etc.).
We would like to complete the integration by ingesting this key to influx.

rh-codebase, 2022-01-31:
etcd2db has been taught to listen for /mon/status which will include /mon/status/<N>, N=1...n. How often are payloads pushed into etcd? After a few minutes, I didn't see any such packets arriving.

caseyjlaw, 2022-01-31:
Thanks!
I had not started a process to run the update, but just started it. It is a screen session "statusmon" on dsa-storage for now.

caseyjlaw, 2022-01-31:
It is on a 10 second cadence. I set up the bottom panel of the correlator dashboard to show the "status" value.

rh-codebase, 2022-01-31:
verified statusmon table is being populated:

rh-codebase, 2022-01-31:
''

caseyjlaw, 2025-02-22:
Is something required to run `etcd2db` for influxdbservice.pro.pvt?  

I believe `statusmon` is no longer being populated after the move of the influx service to the "pro.pvt" network.

E.g.,

```
influx = DataFrameClient('influxdbservice.pro.pvt', 8086, 'root', 'root', 'dsa110')
influx.get_list_measurements()
Out[12]: 
[{'name': 'antmon'},
 {'name': 'bebmon'},
 {'name': 'calmon'},
 {'name': 'corrmon'},
 {'name': 't2mon'},
 {'name': 'wxmon'}]
```

What is required to set up the