dsa110/dsa110-issues #144: webUI monitor point table wonky

caseyjlaw, 2022-02-11:
The DSA-110 WebUI displays a table of monitor points below the control fields.
Currently, it displays values with bad times and other odd values. Here's an example:
[image]

rh-codebase, 2022-02-11:
that's the SNAP table. Has anyone changed the SNAP monitor data?

rh-codebase, 2022-02-11:
Here's the latest monitor packet in etcd for /mon/snap/1:
`{"prog": true, "init": true, "number": 1, "armed_mjd": 59612.68806712963, "time": 59612.68806712963}`

The process sending SNAP monitor data is likely broken or not running.

caseyjlaw, 2022-02-11:
Thanks for digging in, Rick.
Maybe this is a question for @VR-DSA. Is the process running the SNAPs still sending monitor points to etcd?

rh-codebase, 2022-02-11:
Also, that packet looks to have different keys then what etcd2db is expecting. I'll double check but I think there should be a 'snap_num' key instead of 'number' and a 'host' field. I'll dig up what etcd2db and the webui are expecting.