dsa110/dsa110-issues #187: Calibration is not triggered

danasimard, 2022-07-05:
Unsure if preprocessing is not sending the trigger, or if calibration is not receiving it.

danasimard, 2022-07-06:
This seems to be on the preprocessing side of things - there are no messages for calibration triggers in the recent logs.  I've added some logging during the assessments of scans to try to debug this.

danasimard, 2022-07-11:
The issue is that preprocessing falls behind due to the delay added to the etcd callback function in order to account for the files being moved from the staging directory to the correlator directory.

danasimard, 2022-07-11:
@rh-codebase this is where the callback function is defined: https://github.com/dsa110/dsa110-calib/blob/327739cd4542837c22a38ef45e8f6b88d3d14feb/services/preprocess_service.py#L59

A watch is started here: https://github.com/dsa110/dsa110-calib/blob/327739cd4542837c22a38ef45e8f6b88d3d14feb/services/preprocess_service.py#L252 and this script is run as a systemd user service. 

We added a 30s delay because @VR-DSA estimated it takes ~30s for the files to be copied from the staging area to the directory with all of the correlator files. We could remove the delay here, and just check later that the files have arrived in the correlator directory.

rh-codebase, 2022-07-11:
The 30s (ad-hoc) delay is fragile. I'm not keen either of 'checking' since it turns a command(callback) based paradigm into a polling one. It seems like the process which is doing the file copy from the staging area is the one which should have the responsibility of setting "/cmd/cal" command/value. The callback also simply adds the filename to the queue assuming it has been written after the 30sec. What if there was an error in writing/transferring? The writing process knows this(or has the best chance of knowing this), and hence should only send the command when it knows the file  has successfully been written. 

 More generally, producers are responsible for producing content for consumers and letting consumers know the content is ready to be consumed along with any meta-data needed for the consumption. If errors occur, those can be sent as well to reduce the burden on the consumer's error checking logic.

danasimard, 2022-07-11:
That was the original paradigm we had for this, but we changed it when we added the staging of the files on the ssd. Vikram, can the service that moves the file to the nfs change the key when the file is moved?

VR-DSA, 2022-07-12:
It totally can, if you'd like! Right now it's a really simple pull script running on h23, actually, because I wanted to take the load off the corr nodes to see if that helped. It didn't make a difference. In any case, the script is at /dataz/dsa110/operations/copy_files.bash . If you'd like, you could just reinstate the functionality you had in the old preprocessing script to pull the data? It could run on the calibration container - I think we set up the ssh keys? Otherwise, I'm happy to do something with keys on the corr nodes, and use the nfs

danasimard, 2022-07-12:
Okay, I've reverted back to using the rsync handled by the calibration for now.  It's not a general solution, but I think it's fine for now.  

Is it right that the T1 and T3 data copies are polling for new files Vikram?  And both T1 and T3 files are also being staged? Down the line, if we change those to also use callbacks, we should look at a general solution that works for the hdf5 files as well.