dsa110/dsa110-issues #189: adding /mon/sefd key ingest to influxdb

caseyjlaw, 2022-07-15:
I'd like to add a monitor point to the grafana dashboard for system health. This is related to the measurement of a characteristic sensitivity over the whole array. I don't think that fits within any influx keys we use thus far, so I suggest creating one called `/mon/sefd/1`, which has a payload as a python dict with values to be plotted in grafana.
Could you ingest that key, as for others (e.g., `/mon/statusmon/1`).

The python function to push it is given below. Comments welcome.

```
def push_sefd(time0, noise_st, noise_bf, noise_ou):
    """ Set etcd key with sefds                                                                                                                
    """

    version = 1
    sefd_num = 1
    dd = {'sefd_num': sefd_num, 'time': time0.mjd, 'version': version,
	  'noise_st': noise_st, 'noise_bf': noise_bf, 'noise_ou': noise_ou}

    de.put_dict(f'/mon/sefd/{sefd_num}', dd)
```

rh-codebase, 2022-07-15:
I don't see any issues with this. Is SEFD measured for each antenna? If so, if that's something that might be added later then mabe the form: /mon/sefd/<num> where <num> = 1-N for antennas and 0 for entire array. This would line up with how the /mon/ant is logically constructed. One should then be able to join the sefd table with the antenna table if for example plotting sefd vs elevation or some other antenna quantity.  Or does sefd_num represent something entirely different?

caseyjlaw, 2022-07-15:
It isn't structured that way, so I think we need a new one. This is a set of three values that summarize sensitivity in different ways:
```
> ds.get_dict('/mon/sefd/1')
{'sefd_num': 1,
 'time': 59775.477372685185,
 'version': 1,
 'noise_ct': 11959.578125,
 'noise_bf': 10805.8125,
 'noise_ou': 10476.875}
```

They are really "all array" measurements, since the values are taken from subsets of baselines.
If you are ok with having a new "sefdmon" service, then I think we just need the influx connection. Things are all set on the etcd side.

rh-codebase, 2022-07-15:
Those 3 field names would have to be defined as tags which is just a config file change. Then grafana could label them similar to how ant_num is done. The other approach is to create 3 separate queries within the same plot and label each one manually. If you want tags, just pass along the field names which should be defined as such.

caseyjlaw, 2022-07-16:
Ok, I think using tags makes sense. I also wanted to make a change to the names to be more clear.
The three to tag are "noise_short", "noise_beamformer", "noise_long".

rh-codebase, 2022-07-16:
done.

caseyjlaw, 2022-07-16:
I'm not sure what is wrong, but the grafana query stopped returning values for times after your change. I see "noise_short" and other values before that, though. Do I need to change something on the grafana side?

rh-codebase, 2022-07-18:
The latest entry in etcd is:
{"sefd_num": 1, "time": 59778.62464120371, "version": 1, "noise_short": 10517.0390625, "noise_beamformer": 9761.814453125, "noise_long": 9844.98828125}

It is not updating so is the service down?

Influx has the following for the tag "noise_long":

> select "noise_long" from sefdmon
name: sefdmon
time                noise_long
----                ----------
1657940116999999761 9262.21484375
1657940724000000238 9415.109375
1657941329000000238 9312.013671875
1657941934000000238 9432.8603515625
1657942539999999761 9488.326171875

caseyjlaw, 2022-07-18:
The latest in etcd is fine. It stopped today and I restarted it.
The only values showing up in grafana are from two days ago. That coincided with my suggestion to change the tagging scheme. I suspected something broke there, but it isn't clear how from what I see for the panel set up in grafana. Can you inspect that and tell me if I've misconfigured something?

rh-codebase, 2022-07-18:
The latest entries in Influx are ~2 days old as shown in the last comment. The latest entry in etcd at MJD: 59778.62464120371 did not make it into Influx. I'll see what I can find.

rh-codebase, 2022-07-18:
I see the issue. "sefd_num" is still a tag so Influx creates a series using this for uniqueness. The series label to query against are:
noise_beamformer_1
noise_long_1
noise_short_1

here are all the iterations injected into Influx sefdmon measurement:

> select * from sefdmon
name: sefdmon
time                noise_beamformer noise_beamformer_1 noise_bf         noise_ct         noise_long      noise_long_1 noise_ou         noise_short      noise_short_1 sefd_num version
----                ---------------- ------------------ --------         --------         ----------      ------------ --------         -----------      ------------- -------- -------
1657896902999999761                                     10618.55078125   11546.75                                      10219.0517578125                                1        1
1657897509000000000                                     11202.970703125  13366.451171875                               10123.51953125                                  1        1

Influx may have done this because a series was already in the measurement prior to the tag update. I'd like to drop the measurement and start clean to see how the series are labeled. Do you still need sefd_num a tag?

caseyjlaw, 2022-07-18:
Thanks, Rick. We do not need the old tag. Dropping the table seems like the
right thing to do.

rh-codebase, 2022-07-18:
ok. dropping the table and removing sefd_num as a tag. will let you know when that's complete.

rh-codebase, 2022-07-18:
sefdmon table dropped and etcd2db restarted w/out tag: sefd_num. You may need to restart or we'll wait for the next updates.

caseyjlaw, 2022-07-18:
I can confirm that the etcd looks fine and the service is working.
However, I can only get the grafana panel to display if I do `select(*)`, which ends up showing data I don't want. I guess I'm still not understanding tags. Can you make the panel display only the three fields starting with `noise_`?
The dashbaord is running on h23 at http://localhost:3000/d/EXo2BqTGz/correlator?orgId=1&refresh=30s&from=1658168649363&to=1658179449363.

rh-codebase, 2022-07-18:
I forgot about the restrictions for tags. They are string(int) based so they won't preserve a floating point. I'm pretty sure this isn't what you want.. I'm going to remove the tag spec on those values. If you want all 3 on the same graph, then maybe  using a 'like' declaration will work. First step is to get the data into Influx as floats which I'll do now.

rh-codebase, 2022-07-18:
values in the sefdmon measurement now look like:
> select * from sefdmon
name: sefdmon
time                noise_beamformer noise_long   noise_short     sefd_num version
----                ---------------- ----------   -----------     -------- -------
1658182145000000238 7120.7958984375  6424.2578125 8047.6337890625 1        1

I can take a look at that panel later today or tomorrow.

caseyjlaw, 2022-07-19:
Ok, since that change, the grafana plot looks perfect.

closed 2022-07-19 by rh-codebase