dsa110/dsa110-issues #198: tag and update to myrepos

caseyjlaw, 2022-10-27:
I sent an email, but I know Rick prefers to do this properly, so...
I am now happy with calling these tagged versions ready:

-  dsa110-T2 2.8.9
- dsa110-voltage-imaging 0.1.0
- dsa110-event 0.6.12
- dsa110-pyutils v3.7.0-rc1 (I would like to merge my branch into this, though)
- burstfit 0.3.0
- dsa110-cnf v2.0.2
- dsa110-T3 v1.4.6
- dsa110-antpos v1.4.4

There is a release candidate for dsautils. Ideally, you would merge that into main, then try to merge and tag my changes on top of that.
After all that, let’s check in with Vikram on a good time to test deployment.

rh-codebase, 2022-10-28:
origin/development and origin/cjl/dev have already merged work based off of v3.7.0-rc1
There are a few  branches which have not been merged. Are they still under development?
origin/calibration (Dana), origin/dsastorag (dsa-storage-user) and origin/noavg (dsa-storage-user). 
Diffs between v3.7.0-rc1 and origin/development() look to be all backward compatible except:
def get_DM(mjd, ibeam, full):  where 'full' has been added. If this is given a default value, then it will be backward compatible. Its use is only for I/O so this seems worth doing. We can then tag (e684ff2) v3.8.0-rc1 and test. If successful, merge onto master branch and tag v3.8.0

caseyjlaw, 2022-10-29:
I think the other branches do not need to be merged. I saw the development branch `get_DM` function did have a default for that function, so I think it was fine.
I merged development into master and tagged it v3.8.0-rc1. Could you take a look and tag it 'v3.8.0' if you are happy?
It would be great to get this all into myrepos.

rh-codebase, 2022-10-31:
In cli.py the following 2 functions have been defined, then redefined:

def snap(snapnum)
def snap(snapnum, prop, val):
def corr(command):
def corr():

It's unclear what the decorators, mon.command and con.command are doing so it's possible this is ok but it doesn't play nice with pylint.

I'm also not getting unit tests to pass but maybe my env.

caseyjlaw, 2022-10-31:
Ah, yes. The decorators define the way the command-line tools are created. These are different tools to either monitor or command the snaps, so the names are appropriate.

rh-codebase, 2022-10-31:
are you passing unit tests in dsautils?

rh-codebase, 2022-10-31:
yes. I'm not sure what is going on in dsa110-shell. I see a lot of 3.0.0-rcXXX tags but they seemed to have blown by a 3.1.0-rc1 tag. I would normally update the .mrconfig_production with all the rc tags, test, then label all the rc tags with non rc(ie. v3.8.0-rc1 -> v3.8.0 and then update .mrconfig_production. There's no churn on tagging so should be safe.