dsa110/dsa110-issues #286: Unable to reach grafana dashboard and query influxdb

mbsherma, 2024-04-22:
I am currently unable to reach the grafana dashboard or query from influxdb. By my understanding, they should be reachable via dsa110maas or dsastorage, which I am able to reach by ssh over the ovro network (e.g. dsastorage.ovro.pvt and dsa110maas.ovro.pvt). When I tried using 'ping grafanaservice.sas.pvt  and ping influxdbservice.sas.pvt',  I get the error 'Destination Host Unreachable'. Are they reachable over another network? I'm aware this may just be due to the changes during commissioning and become more stable later on.

Thanks!

kembyct, 2024-04-22:
I think you should get on lwamaas system in order to access those. It's a separate gateway.

mbsherma, 2024-04-22:
@kembyct Ok, I don't think I have access to lwamaas, what hostname should I use to ssh there? And which ports are used for grafana, etc.? Currently I have this for dsa110maas in my .ssh/config, so I assume its something similar:

Host dsa110maas
    user ubuntu
    hostname dsa110maas.ovro.pvt
    ProxyJump ovro
    LocalForward 3000 grafanaservice:3000
    LocalForward 5008 dsa-storage.ovro.pvt:5008
    LocalForward 5240 localhost:5240
    LocalForward 9090 webserveruiservice:9090

mbsherma, 2024-04-22:
@rh-codebase @kembyct Vikram informed me that h00 is currently refusing to boot, and that is likely the cause of this. We can pause on this issue until its back online.

kembyct, 2024-04-22:
OK. But it's reachable from lwamaas. I tried it.

kembyct, 2024-04-22:
I take it back. I've just noticed the grafana on lwa system is seperate from the one on dsa system. I'll look further into this, and get back to you shortly.
Thanks.

mbsherma, 2024-04-22:
Gotcha, Vikram informed me about the down server as well, thanks for taking a look!