dsa110/dsa110-issues #319: Adding /mon/nsfrb/fastvis to etcd

mbsherma, 2025-02-14:
I'd like to add a new etcd key for the realtime nsfrb subsystem that will allow the nsfrb correlator ([dsaX_nsfrb](https://github.com/dsa110/dsa110-xengine/blob/63e4ffc18a9faaf04cf765df5b085ad46ad7ff10/src/dsaX_nsfrb.c)) to alert the [realtime imager](https://github.com/dsa110/dsa110-nsfrb/tree/development/realtime) when data is ready and where it is memory mapped. The payload would be a python dict with fields 'ready' (bool, 0 or 1) and 'shmid' (int). Can we add this to etcd?

rh-codebase, 2025-02-14:
You sure can. There's no need to define keys ahead of time. IIRC the DStore(?) class could be used here. If you need this information into Influx/Grafana, then give a shout. That part is not automatic.

mbsherma, 2025-02-14:
Oh ok I see, In that case I'll go ahead and try DStore here and reach out again for when I need to add grafana keys. Thanks!